laurb9/PanoController-Firmware #7: Remember vertical orientation

laurb9, 2016-03-05:
Vertical carrier orientation is relative to platform but as there is no end marker, calibration is not possible. Do not want to keep motors on all the time, but it may be what needs to be done.

CarriedBy6, 2022-07-15:
Might want to consider an absolute position sensor that way you don’t necessarily need end stops or homing procedures. [AS5048A](https://ams.com/as5048a) I added them to my axis and found it to be enough resolution (16,000+ positions at 360deg) even at 600mm.

laurb9, 2022-07-24:
This is a good idea, thanks. I didn't know these existed. A Hall effect or optical sensor is more cost effective if a 9-axis MPU is also installed, but this position sensor has potential.