laurb9/PanoController-Firmware #9: Add radio transmitter (opt)

laurb9, 2016-03-05:
- All of the display and input work could be offloaded to handheld radio remote device but the system is no longer self-contained.

laurb9, 2016-04-03:
I've identified an affordable RF board that could work: NRF24L01 2.4Ghz
http://arduino-info.wikispaces.com/Nrf24L01-2.4GHz-HowTo

Defragster, 2016-04-09:
Started at the top just seeing this - I'm using ESP8266 under Arduino with my Teensy, pretty effective - full web server support with it's own processor on par with the Teensy for under $5.

laurb9, 2016-04-09:
Some interactive features such as camera positioning would be adversely affected by WiFi latency. Others, like camera settings setup, would benefit from the nicer UI in web page. Unfortunately, I do need the interactive part more than the other, because with the camera on a pole it is hard to see what it's doing and intervene.

What I'm thinking here is move the menu and controls to a detachable remote piece. This increases the complexity and needs two microcontroller boards, but offloading the display work allows the motor controller side to do more.

laurb9, 2016-08-29:
Implemented a simple proof of concept which synchronizes inputs across two units via NRF24L01. For example, IR and Bluetooth remote inputs from one device are blindly sent to the unit containing the driver.

Syncing the display should be next.

laurb9, 2016-10-24:
Abandoned the above-mentioned proof of concept as it had to future path, and instead implemented #58 with a smart remote control unit with menus using the NRF24.

laurb9, 2016-10-24:
Bluetooth integration (#10) should be relatively easy to build on top of this.

closed 2016-10-24 by laurb9