laurb9/PanoController-Firmware #47: Real Time Clock

laurb9, 2016-04-03:
Add an RTC. 
Options are:
- a standalone DS3231 (works with any Arduino), or 
- add a CR16 battery + crystal to the teensy board

Possible uses:
- display timestamp on pano info screen, so a screenshot can be used to match against photo sets.
- record panos taken in eeprom, and dump them via serial later.

Defragster, 2016-04-09:
ESP8266 to your phone would have NTP server access - except when you go out in the wilds without cell service - and turn off the ESP8266 to save battery.

I've put a couple crystals on my Teensy's and even wired one to a CR2032 once.

laurb9, 2016-04-09:
Auto setting time would be a nice feature, maybe not worth adding the power hungry ESP just for it, but the other capabilities are worth exploring.

The date doesn't have to that accurate, it only has to match the camera timestamp.
Mulling over a GPS unit. There are a few cheap ones out there, not sure how good.

laurb9, 2017-08-14:
Could add capability to sync time via BLE (after #10 is done)

laurb9, 2017-10-02:
This is less important now that the app keeps all the info.

closed 2017-10-15 by laurb9