laurb9/PanoController-Firmware #48: Pano log in EEPROM

laurb9, 2016-04-03:
Record information for each panoramas taken in EEPROM.
Dump the EEPROM over serial when connected to PC, for identifying the photo sets and settings easier.

Defragster, 2016-04-09:
The $8 PJRC prop shield has 8MB Flash and a HID software transfer app.

But an ESP8266 could save them to onboard 3MB SPIFFS and also post them in real time to the control web page to cut and paste to a file - or save the file for FTP transfer when done.

Recording the camera IMAGE number will make it easier to associate the notes - even if using a fresh flash card for each set.

laurb9, 2016-04-09:
Yes, recording more detail such as image numbers with timestamps for each photo is a good idea, thanks. Maybe elevation and azimuth also, if I add a compass could help with prealignment.

These aren't needed in real time though, since all the stitching happens at home base anyway.

laurb9, 2017-10-02:
May not need to be in EEPROM at all if it can be transferred to the app immediately.
Logging the position, timestamps and other known metadata for each photo is still useful.