laurb9/PanoController-Firmware #50: IR shutter remote

laurb9, 2016-04-09:
Add an IR transmitter to trigger the shutter.

# simpler: no more shutter cable to plug and unplug and expose opening to dust
# on A6000, IR remote can trigger a full bracketing sequence in one quick press, whereas the wired shutter needs to be held down the entire time otherwise

possible problems:
# strong sunlight issues ?

Defragster, 2016-04-09:
Or use an ESP8266 or similar.  Do data entry and control from a web page on your cell phone or tablet, buy more batteries.

laurb9, 2016-04-09:
This is to replace the teensy-to-camera shutter cable. The controller uses this to take a photo (or more) at each position, autonomously. Since most DSLRs have IR sensor, if I put the IR LED on the platform and use it to trigger the shots, I can get rid of a dangling (and possibly expensive) cable.

Defragster, 2016-04-09:
My camera like yours needs the press to hold for bracketing to complete - Does an IR interface deal with that or does I maintain a press?

laurb9, 2016-04-10:
At least for my camera, the IR remote needs a single quick press, which is the main reason I kind of want it. I've miscalculated the shutter timing a few times and I ended up having two instead of three pics in each frame.

Defragster, 2016-04-10:
I got a wired remote - not used it for bracketing.  So silly you select bracket - forget to hold the button - or forget to turn of bracketing - then until you do - alternate pictures are underexposed/overexposed ... or just right. If the IR can say 'takes pics' - not sure why the button doesn't. Getting camera feedback would close the loop