laurb9/PanoController-Firmware #54: flash socket for shutter feedback ?

laurb9, 2016-04-11:
Investigate if it's possible to use the flash hotshoe for shutter feedback.

This allows figuring everything out from a test shot.

CarriedBy6, 2022-07-15:
I’m in the process of designing a gigapixel head myself. I have had luck creating a closed loop system using a Sony A7RIII and a Vello hot shoe adapter like this [Vello Adapter](https://www.bhphotovideo.com/c/product/1349065-REG/vello_hsa_smpcs_hsa_for_sony_multi.html) in order to detect when the exposure has finished and signal the head that it’s ok to move. It protects against a motion blurred pano having forgotten to match camera and head exposure duration in the menus. It’s also a big time saver for HDR. It gives you tighter timings instead of wasting time during shorter exposures having to account for your longest exposure in a set. I agree it would be great to also get the lens focal length settings automatically but my fix is a simple interrupt edge detection rather than decoding the Sony flash protocol.

laurb9, 2022-07-24:
This issue was indeed about the standard flash "switch", the other one you are referring to about decoding protocols is probably #55. Did you find that the duration matches the exposure time ?